aliasrobotics/RVD #10: Relative Path Traversal vulnerability in SREA-01 and SREA-50

aliasbot, 2018-08-07:
| Input    | Value  |
|----------|--------|
| Robot Component | SREA-01 and SREA-50  |
| Vendor | ABB  |
| CVE ID | N/A  |
| CWE ID | Path Traversal (CWE-22)  |
| RVSS Score  | 10.0  |
| RVSS Vector | RVSS:1.0/AV:RN/AC:L/PR:N/UI:N/Y:T/S:U/C:L/I:H/A:H/H:N |
| GitHub Account | aliasbot |
| Date Reported | 2017-02-07 |
| Date Updated | N/A |
| Exploitation vector | Relative Path Traversal |
 
### Description:
 Relative Path Traversal vulnerability in SREA-01 and SREA-50 legacy remote monitoring tools and Ethernet adapters in ABB could access files on the affected products&acirc; file systems, view data, change configuration, retrieve password hash codes, and potentially insert and send commands to connected devices without authorization via HTTP request which refers to files using ../../ relative paths.

On versions: SREA-01 revisions A, B, C: application versions up to 3.31.5 - SREA-50 revision A: application versions up to 3.32.8

###### Credits to: Bertin Jose and Fernandez Ezequiel

vmayoral, 2019-12-08:
Review dates in this ticket. Used the following sources:
- https://www.us-cert.gov/ics/advisories/ICSA-17-222-05 (mitigation)
- https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2017-9664 (report, detection)

glerapic, 2020-05-20:
Ticket Updated.