aliasrobotics/RVD #1405: [sound_play/scripts/soundplay_node.py] fix Gstreamer memory leak

glerapic, 2020-03-20:
`From This pr: https://github.com/ros-drivers/audio_common/pull/126`

`stop()` function in class `soundtype` seems not to free Gstreamer objects correctly.
Gstreamer object has to be unreferenced explicitly (cf. [Gst.Element.get_bus](https://lazka.github.io/pgi-docs/Gst-1.0/classes/Element.html#Gst.Element.get_bus)), otherwise causes a memory leak and leads the node to crash. #109
 
I tried to free the Gstreamer objects by substituting `None`.

```yaml
{
    "id": 508,
    "title": "rcl_action: data race, eprosima::fastrtps::rtps::Participant...",
    "type": "bug",
    "description": "Issue detected while running Google Sanitizers.\n\n ",
    "cwe": "None",
    "cve": "None",
    "keywords": [
        "components software",
        "master",
        "package: rcl_action",
        "robot component: ROS2",
        "weakness"
    ],
    "system": "ros2",
    "vendor": "N/A",
    "severity": {
        "rvss-score": "None",
        "rvss-vector": "N/A",
        "severity-description": "",
        "cvss-score": 0,
        "cvss-vector": ""
    },
    "links": [
        "https://github.com/aliasrobotics/RVD/issues/508"
    ],
    "flaw": {
        "phase": "testing",
        "specificity": "ROS-specific",
        "architectural-location": "platform code",
        "application": "N/A",
        "subsystem": "cognition:ros2",
        "package": "rcl_action",
        "languages": "None",
        "date-detected": "Mon, 21 Oct 2019 07:39:17 +0000",
        "detected-by": "",
        "detected-by-method": "testing dynamic",
        "date-reported": "Mon, 21 Oct 2019 07:39:17 +0000",
        "reported-by": "Alias Robotics (http://aliasrobotics.com)",
        "reported-by-relationship": "automatic",
        "issue": "https://github.com/aliasrobotics/RVD/issues/508",
        "reproducibility": "always",
        "trace": null,
        "reproduction": "Find a    pre-compiled environment in the Docker image below. Reproducing it implies    source the workspace, finding the appropriate test and executing it.",
        "reproduction-image": "registry.gitlab.com/aliasrobotics/offensive/alurity/ros2/ros2:build-tsan2-commit-b2dca472a35109cece17d3e61b18af5cb9be5772"
    },
    "exploitation": {
        "description": "",
        "exploitation-image": "",
        "exploitation-vector": ""
    },
    "mitigation": {
        "description": "",
        "pull-request": ""
    }
}
```

LanderU, 2020-03-20:
@glerapic could you please add the labels for this one?

LanderU, 2020-03-20:
Also, the syntax is incorrect, should be RVD#1405 ...

vmayoral, 2020-03-20:
thanks guys, another tiny request, @glerapic, can we review the `title` and also adapt the upper one?

vmayoral, 2020-04-02:
Content removed from upper comment:

> `From This pr: https://github.com/ros-drivers/audio_common/pull/126`
>
>
> `stop()` function in class `soundtype` seems not to free Gstreamer objects correctly.
> Gstreamer object has to be unreferenced [...]

closed 2020-11-24 by rvd-bot