aliasrobotics/RVD #1406: Felix shell console access without credentials on port 6666 (default)

LanderU, 2020-03-30:
```yaml
{
    "id": 0,
    "title": "Felix shell console access without credentials on port 6666 (default)",
    "type": "vulnerability",
    "description": "We found that the Universal Robots Controllers has Felix Shell console application enabled on port 6666 (default). By netcat connection anyone can perform any of the several actions Felix Shell console allows to users (such as shutdown). Exemplary commands can be found at https://portal.liferay.dev/docs/7-0/reference/-/knowledge_base/r/using-the-felix-gogo-shell",
    "cwe": null,
    "cve": null,
    "keywords": [
        "Universal Robots",
        "manipulation",
        "cobot",
        "CB 3.1",
        "CB 3.4.5"
    ],
    "system": "Universal Robots Robot Controllers CB 3.10, 3.11, 3.12, 3.12.1",
    "vendor": "Universal Robots",
    "severity": {
        "rvss-score": 9.5,
        "rvss-vector": "RVSS:1.0/AV:AN/AC:L/PR:N/UI:N/Y:Z/S:U/C:L/I:H/A:H/H:E",
        "severity-description": "critical",
        "cvss-score": 8.3,
        "cvss-vector": "CVSS:3.0/AV:A/AC:L/PR:N/UI:N/S:U/C:L/I:H/A:H/"
    },
    "links": [
        "https://portal.liferay.dev/docs/7-0/reference/-/knowledge_base/r/using-the-felix-gogo-shell",
        "https://felix.apache.org/documentation/subprojects/apache-felix-remote-shell.html",
        "https://felix.apache.org/documentation/subprojects/apache-felix-gogo.html"
    ],
    "flaw": {
        "phase": "tested",
        "specificity": "subject-specific",
        "architectural-location": "application-specific code",
        "application": "manipulator, control box",
        "subsystem": "cognition:manipulation",
        "package": "Felix Gogo Apache",
        "languages": "shell",
        "date-detected": "2020-1-28",
        "detected-by": "Unai Ayucar Carbajo, Endika Gil Uriarte, Alfonso Glera Picon, V\u00edctor Mayoral Vilches, Xabies Saez de C\u00e1mara, Lander Usategi San Juan (Alias Robotics)",
        "detected-by-method": "testing violation",
        "date-reported": "2020-03-30",
        "reported-by": "Unai Ayucar Carbajo, Endika Gil Uriarte, Alfonso Glera Picon, V\u00edctor Mayoral Vilches, Xabies Saez de C\u00e1mara, Lander Usategui San Juan (Alias Robotics)",
        "reported-by-relationship": "security researcher",
        "issue": null,
        "reproducibility": "always",
        "trace": "Not disclosed",
        "reproduction": "Not disclosed",
        "reproduction-image": "Not disclosed"
    },
    "exploitation": {
        "description": "Not disclosed",
        "exploitation-image": "Not disclosed",
        "exploitation-vector": "Not disclosed"
    },
    "mitigation": {
        "description": "Not disclosed",
        "pull-request": "Not disclosed",
        "date-mitigation": null
    }
}
```

LanderU, 2020-03-30:
alurity.yml used to reproduce this:

```yaml
networks:
  - network:
    - driver: overlay
    - name: urnetwork
    - encryption: false

containers:
  - container:
    - name: ur_31
    - modules:
         - base: registry.gitlab.com/aliasrobotics/offensive/alurity/robo_ur_cb3_1:3.12.1
         - network: urnetwork
   - container:
     - name: ur_311
     - modules:
          - base: registry.gitlab.com/aliasrobotics/offensive/alurity/robo_ur_cb3_1:3.11
          - network: urnetwork
   - container:
     - name: ur_312
     - modules:
         - base: registry.gitlab.com/aliasrobotics/offensive/alurity/robo_ur_cb3_1:3.12
          - network: urnetwork
  - container:
    - name: ur_3121
    - modules:
         - base: registry.gitlab.com/aliasrobotics/offensive/alurity/robo_ur_cb3_1:3.12.1
         - network: urnetwork
  - container:
    - name: attacker
    - modules:
         - base: registry.gitlab.com/aliasrobotics/offensive/alurity/alurity:latest
         - volume: registry.gitlab.com/aliasrobotics/offensive/alurity/expl_robosploit/expl_robosploit:latest
         - volume: registry.gitlab.com/aliasrobotics/offensive/alurity/deve_atom:latest
         - volume: registry.gitlab.com/aliasrobotics/offensive/alurity/reco_nmap:latest
         - network: urnetwork
```

vmayoral, 2020-03-30:
Exploitation demonstrated at https://www.youtube.com/watch?v=tS2NpgHpz_0&feature=youtu.be

vmayoral, 2020-04-01:
ping @LanderU, can we get a CWE here? Also, we should get a CVE ID for this one.

LanderU, 2020-04-01:
Added CWE.