aliasrobotics/RVD #1443: UR dashboard server enables unauthenticated remote control of core robot functions

bedieber, 2020-04-01:
Fill in following the example below. If you need further clarifications on any of the items, refer to our [taxonomy](https://github.com/aliasrobotics/RVD/blob/master/docs/TAXONOMY.md) (remove these lines line).

```yaml
id: 
title: "UR dashboard server enables unauthenticated remote control of core robot functions"
type: vulnerability
description: "Universal Robots Robot Controllers Version CB2 SW Version 1.4 upwards, CB3 SW Version 3.0 and upwards, e-series SW Version 5.0 and upwards expose a service called DashBoard server at port 29999 that allows for control over core robot functions like starting/stopping programs, shutdown, reset safety and more. The DashBoard server is not protected by any kind of authentication or authorization."
cwe: "CWE-306 (Missing Authentication for Critical Function)"
cve: ""
keywords: [
    "Universal Robots",
    "manipulation",
    "cobot",
    "CB 3.1",
    "CB 3.4.5",
    "CB 2", 
    "e-series"
]
system: "Universal Robots Robot Controllers CB 2, CB3, e-series"
vendor: "Universal Robots"
severity:
  rvss-score: 0
  rvss-vector: ""
  severity-description: "critical"
  cvss-score: 
  cvss-vector: ""
links: [
    'https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/dashboard-server-e-series-port-29999/',
    'https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/dashboard-server-cb-series-port-29999/',
]
flaw:
  phase: testing
  specificity: subject-specific
  architectural-location: application-specific code
  application: manipulator, control box
  subsystem: cognition:manipulation
  package: N/A
  languages: N/A
  date-detected: 
  detected-by: 
  detected-by-method: testing violation
  date-reported: 2020-04-01 (15:00)
  reported-by: Bernhard Dieber, Benjamin Breiling (and many others)
  reported-by-relationship: security researcher
  issue: 
  reproducibility: always
  trace: N/A
  reproduction: Not disclosed
  reproduction-image: Not disclosed
exploitation:
  description: Not disclosed
  exploitation-image: Not disclosed
  exploitation-vector: Not disclosed
mitigation:
  description: Not disclosed
  pull-request: Not disclosed
```

bedieber, 2020-04-01:
That is a well-known "function" of the UR controller, cannot say for sure who actually "discovered" it or where it was first reported as security-related flaw

unaithetutamatumatu, 2020-04-01:
I would love to add the CSS vector:
CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:U/C:L/I:H/A:H
with 9.4 scoring! WOW!

and RVSS vector aswell:
RVSS:1.0/AV:RN/AC:L/PR:N/UI:N/Y:O/S:U/C:L/I:H/A:H/H:E
score 10!

Really critical one!

vmayoral, 2020-04-01:
Thanks for the assessment @unaithetutamatumatu. @bedieber can you confirm you agree with the criticality evaluation @unaithetutamatumatu proposes? Refer to https://github.com/aliasrobotics/RVSS if you need to do further readings on the vectors.

bedieber, 2020-04-01:
> I would love to add the CSS vector:
> CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:U/C:L/I:H/A:H
> with 9.4 scoring! WOW!
> and RVSS vector aswell:
> RVSS:1.0/AV:RN/AC:L/PR:N/UI:N/Y:O/S:U/C:L/I:H/A:H/H:E
> score 10!
> Really critical one!

Yes I can confirm those scores.

LanderU, 2020-04-01:
We've created and alurity.yml to validate this scenario:

```bash
networks:
  - network:
    - driver: overlay
    - name: urnetwork
    - encryption: false

containers:
  - container:
    - name: ur_3121
    - modules:
         - base: registry.gitlab.com/aliasrobotics/offensive/alurity/robo_ur_cb3_1:3.12.1
         - network: urnetwork
  - container:
    - name: attacker
    - modules:
         - base: registry.gitlab.com/aliasrobotics/offensive/alurity/alurity:latest
         - volume: registry.gitlab.com/aliasrobotics/offensive/alurity/expl_robosploit/expl_robosploit:latest
         - volume: registry.gitlab.com/aliasrobotics/offensive/alurity/deve_atom:latest
         - volume: registry.gitlab.com/aliasrobotics/offensive/alurity/reco_nmap:latest
         - network: urnetwork
```

LanderU, 2020-04-01:
Also, our team has developed a robosploit module to validate it, you can check it here:

https://www.youtube.com/watch?v=FBiASTrPzCw&feature=youtu.be

bedieber, 2020-04-02:
> Also, our team has developed a robosploit module to validate it, you can check it here:
> https://www.youtube.com/watch?v=FBiASTrPzCw&feature=youtu.be

very nice work!

vmayoral, 2020-04-04:
Assigned CVE ID CVE-2020-10265, thanks for the contribution!