aliasrobotics/RVD #1444: RVD#1444: RTDE Interface allows unauthenticated reading of robot data and unauthenticated writing of registers and outputs

bbreilin, 2020-04-01:
Fill in following the example below. If you need further clarifications on any of the items, refer to our [taxonomy](https://github.com/aliasrobotics/RVD/blob/master/docs/TAXONOMY.md) (remove these lines line).

```yaml
id: 
title: "RVD#1444: RTDE Interface allows unauthenticated reading of robot data and unauthenticated writing of registers and outputs"
type: vulnerability
description: "CB3 SW Version 3.3 and upwards, e-series SW Version 5.0 and upwards allow authenticated access to the RTDE (Real-Time Data Exchange) interface on port 30004 which allows setting registers, the speed slider fraction as well as digital and analog Outputs. Additionally unautheticated reading of robot data is also possible"
cwe: "CWE-200 (Exposure of Sensitive Information to an Unauthorized Actor)"
cve: 
keywords: [
    "Universal Robots",
    "RTDE",
    "cobot",
    "CB 3.3",
    "e-series"
]
system: "Universal Robots Robot Controllers CB 3.1 3.3 and upwards, e-series 5.0 and upwards"
vendor: "Universal Robots"
severity:
  rvss-score: 0
  rvss-vector: ""
  severity-description: "critical"
  cvss-score: 
  cvss-vector: 
links: [
https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/real-time-data-exchange-rtde-guide/
]
flaw:
  phase: testing
  specificity: subject-specific
  architectural-location: application-specific code
  application: manipulator, control box
  subsystem: cognition:manipulation
  package: N/A
  languages: N/A
  date-detected:
  detected-by: Bernhard Dieber, Benjamin Breiling (and many others)
  detected-by-method: testing violation
  date-reported: 2020-04-01 (16:00)
  reported-by: Bernhard Dieber, Benjamin Breiling (and many others)
  reported-by-relationship: security researcher
  issue: 
  reproducibility: always
  trace: N/A
  reproduction: Not disclosed
  reproduction-image: Not disclosed
exploitation:
  description: Not disclosed
  exploitation-image: Not disclosed
  exploitation-vector: Not disclosed
mitigation:
  description: Not disclosed
  pull-request: Not disclosed
```

vmayoral, 2020-04-01:
@bbreilin a couple of things here while triaging. I couldn't find any details about CB 3.3 and AFAIK, only the following exists:

- CB2: controller type is CB2 robot type with flash card. UR5 and UR10
- CB2.1: controller type is CB2 robot type with USB stick. UR5 and UR10
- CB3.0: controller type is CB3 robot type with flash card. UR5 and UR10
- CB3.1: controller type is CB3 robot type with USB stick. UR3, UR5 and UR10

Could you maybe be a bit more specific on your version descriptions above
([official source](https://www.universal-robots.com/download/?option=66367#section18860))

I'll assume you refer to CB3.1 but let me know otherwise.

vmayoral, 2020-04-01:
Receives the following vector with CVSS 3.0 `CVSS:3.1/AV:A/AC:L/PR:N/UI:N/S:U/C:H/I:H/A:H` and a severity scoring of `8.8`.

vmayoral, 2020-04-01:
Asigned preliminarily CVE-2020-10264.

LanderU, 2020-04-01:
Our team is working on an alurity.yml file to validate this scenario. Also, a robosploit module will be created.

vmayoral, 2020-04-04:
Removed triage label, CVE asignation on its way https://github.com/CVEProject/cvelist/pull/3514. Thanks again @bbreilin and @bedieber for the ticket. 

Your CVE ID will be `CVE-2020-10264`

LanderU, 2020-04-04:
Video that shows the process:

https://www.youtube.com/watch?v=R4y6d2PXfgM

vmayoral, 2020-05-18:
Minor edition in this ticket to avoid https://pyyaml.org/wiki/YAMLColonInFlowContext