aliasrobotics/RVD #1445: RVD#1445: Default password for root user on robot controller

bbreilin, 2020-04-01:
Fill in following the example below. If you need further clarifications on any of the items, refer to our [taxonomy](https://github.com/aliasrobotics/RVD/blob/master/docs/TAXONOMY.md) (remove these lines line).

```yaml
id:
title :RVD#1445: Default password for root user on robot controller
type: vulnerability
description: "Default password for root user on the robot controller"
cwe: "CWE-798 (Use of Hard-coded Credentials)"
cve: "CVE-2018-10633"
keywords: [
    "Universal Robots",
    "password",
    "cobot",
    "CB 2"
    "CB 3.1",
    "e-series"
]
system: "Universal Robots Robot Controllers"
vendor: "Universal Robots"
severity:
  rvss-score: 0
  rvss-vector: ""
  severity-description: "critical"
  cvss-score: 
  cvss-vector: 
links: [
   
]
flaw:
  phase: testing
  specificity: subject-specific
  architectural-location: application-specific code
  application: manipulator, control box
  subsystem: cognition:manipulation
  package: N/A
  languages: N/A
  date-detected: 
  detected-by: Bernhard Dieber, Benjamin Breiling (and many others)
  detected-by-method: online research
  date-reported: 2018-04-01 (16:30)
  reported-by: Bernhard Dieber, Benjamin Breiling
  reported-by-relationship: security researcher
  issue:
  reproducibility: always
  trace: N/A
  reproduction: Not disclosed
  reproduction-image: Not disclosed
exploitation:
  description: Not disclosed
  exploitation-image: Not disclosed
  exploitation-vector: Not disclosed
mitigation:
  description: Not disclosed
  pull-request: Not disclosed
```

vmayoral, 2020-04-01:
Thanks for this @bbreilin :) ! Unfortunately, it seems this was already registered. Please see https://github.com/aliasrobotics/RVD/issues/672. 

I'm closing this ticket for now but let me know if you disagree and you appreciate relevant differences.

closed 2020-04-01 by vmayoral

vmayoral, 2020-05-20:
Added the `invalid` label to exclude it from duplicates analysis.