aliasrobotics/RVD #3315: Cleartext transmission of sensitive information in MAVLink protocol version 1.0 and 2.0

vmayoral, 2020-06-30:
```yaml
id: 
title: 'RVD#2569: Cleartext transmission of sensitive information in MAVLink protocol version 1.0 and 2.0'
type: vulnerability
description:  This vulnerability applies to the Micro Air Vehicle Link (MAVLink) protocol
  and allows a remote attacker to gain access to sensitive information provided it has
  access to the communication medium. MAVLink is a header-based protocol that does 
  not perform encryption to improve transfer (and reception speed) and efficiency by 
  design. The increasing popularity of the protocol (used accross different autopilots) 
  has led to its use in wired and wireless mediums through insecure communication 
  channels exposing sensitive information to a remote attacker with ability to intercept 
  network traffic.
cwe: CWE-319
cve: CVE-2020-10281
keywords:
- MAVLink
- v1.0
- v2.0
- PX4
- Ardupilot
system: MAVLink
vendor: ""
severity:
  rvss-score: 7.3
  rvss-vector: RVSS:1.0/AV:AN/AC:L/PR:N/UI:N/S:U/Y:T/C:H/I:N/A:N/H:N
  severity-description: high
  cvss-score: 7.5
  cvss-vector: CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:U/C:H/I:N/A:N
links:
- https://arxiv.org/abs/1906.10641
- https://arxiv.org/abs/1905.00265
- https://docs.google.com/document/d/1ETle6qQRcaNWAmpG2wz0oOpFKSF_bcTmYMQvtTGI8ns/edit
- https://docs.google.com/document/d/1upZ_KnEgK3Hk1j0DfSHl9AdKFMoSqkAQVeK8LsngvEU/edit
- https://docs.google.com/document/d/1XtbD0ORNkhZ8eKrsbSIZNLyg9sFRXMXbsR2mp37KbIg/edit
flaw:
  phase: unknown
  specificity: subject-specific
  architectural-location: platform code
  application: Flying vehicles and/or others using MAVLink protocol.
  subsystem: communication
  package: N/A
  languages: C, C++
  date-detected: 2020-04-20
  detected-by: 
  detected-by-method: testing
  date-reported: '2020-06-30'
  reported-by: "Victor Mayoral Vilches (Alias Robotics)"
  reported-by-relationship: security researcher
  issue: 
  reproducibility: always
  trace: N/A
  reproduction: N/A
  reproduction-image: N/A
exploitation:
  description: Not available
  exploitation-image: Not available
  exploitation-vector: Not available
  exploitation-recipe: ''
mitigation:
  description: See https://arxiv.org/abs/1905.00265 for a first approach though not source code was found at the time of reporting.
  pull-request: N/A
  date-mitigation: null
```

vmayoral, 2020-06-30:
Likely applying also to other robot components. Ping @glerapic, let me know if you disagree with this ticket, otherwise I'm requesting the CVE ID preliminarily assigned.

glerapic, 2020-07-01:
LGTM!

vmayoral, 2020-07-03:
Assigned a CVE ID, sent a PR to the upstream CVE List repo https://github.com/CVEProject/cvelist/pull/4247